microsoft/pxt-maker #5: Feedback on maker.makecode.com

abchatra, 2018-01-25:
1. The breadboard blends in with the background. I think it's more clear when there's a distinction:
[image]
vs.
[image]

2. The +/- "zoom" buttons in the bottom right only zoom the blocks. I would have expected them to at least zoom the simulator. The breadboard especially is helpful to zoom since there's a lot of detail there (whereas with the microbit simulator, the detail isn't as small.)

darzu, 2018-01-25:
Looks better today, thanks!
[image]

pelikhan, 2018-01-26:
Fixed. Zoom is a deeper issue not addressed here.

closed 2018-01-26 by pelikhan