microsoft/pxt-maker #12: Keyboard blocks cannot accept variable

deanm1278, 2018-01-26:
Trying to do this doesn't work:
[image]

I can get it to work in the javascript editor using
```
let item = 0
loops.forever(function () {
    keyboard.type(pins.A0.analogRead().toString())
})
```

pelikhan, 2018-01-26:
Could you click on "share" and paste the program URL?

deanm1278, 2018-01-26:
https://makecode.com/_H3p3r5VFk8ua

pelikhan, 2018-01-26:
Right, this block only accepts a string and don't auto convert numbers to string. Try 

https://makecode.com/_aR38RrhE0APC

ladyada, 2018-01-26:
that works! but...is super unintuitive :/
i dont know javascript as much as other langs but maybe we can check the type and print out the 'displayable' version? Arduino Serial.print() and python print() do what you expect :)

pelikhan, 2018-01-26:
@riknoll could we have an automatic conversion from number to string in blocks?

pelikhan, 2018-11-05:
Fixed.

closed 2018-11-05 by pelikhan