microsoft/pxt-maker #47: SPI support ?

sabas1080, 2018-05-02:
Hi

I am testing with a sparkfun dev board

I want to try an SPI sensor, but I do not see any examples

I tried with

`spiPins(mosi: DigitalPin, miso: DigitalPin, sck: DigitalPin)`

but it does not look like a valid command

Thanks

pelikhan, 2018-05-02:
@mmoskal I see CODAL_SPI but I don't think we've exposed a typescript Api for it yet.

sabas1080, 2018-05-02:
Is there a plan to do it or approximate date?

pelikhan, 2018-05-02:
Right, we haven't migrated yet those APIs, but it shouldn't be too hard to add them.

https://github.com/Microsoft/pxt-microbit/blob/master/libs/core/pins.cpp#L385

pelikhan, 2018-05-04:
Here's the PR
https://github.com/Microsoft/pxt-common-packages/pull/307

pelikhan, 2018-05-04:
And it's live.

closed 2018-05-04 by pelikhan

sabas1080, 2018-05-04:
Thank you very much

mmoskal, 2018-05-04:
We might be changing transfer API to be full duplex. Best use it with one of the buffers being `null` - this will continue to work as it does now.