microsoft/pxt-maker #48: Serial function available

sabas1080, 2018-05-02:
Hi

The function serial write, read or select pin TX and TX or port 2, available?

I would like to use the serial port 2 of my board

Thanks

pelikhan, 2018-05-04:
@mmoskal let's chat today about serial vs hf2 vs webhf2

mmoskal, 2018-05-04:
@sabas1080 I assume you mean the TX/RX pins, not serial emulated over USB?

mmoskal, 2018-05-04:
@pelikhan we should totally do RX/TX pin stuff - it's already in Codal, same deal as with SPI.

USB serial needs some thought.

pelikhan, 2018-05-04:
Correct. The current terminology is confusing. "serial" goes over HF2 which off course is confusing when you want to do TX/RX serial. Maybe we should renamed the HF2 stuff to just "console"  (and it could spit out to serial as well).

sabas1080, 2018-05-04:
For my is:

CDC vs hf2 vs webhf2 , PC<----> Board, that I hope we have something for "debugging"

I was talking serial in pins TX and RX (or the SERCOM), board <--->sensor

Thanks @pelikhan @mmoskal

closed 2018-07-30 by sabas1080