microsoft/pxt-maker #61: Multiplex SPI in SERCOM

sabas1080, 2018-07-31:
Hi @pelikhan @mmoskal 

**Describe the bug**
I have SPI working, but in the SERCOM4 default pins, how can I change that?

the change in config.ts does not work

How do I choose the SERCOM and PADs?.

**Additional context**
https://github.com/Microsoft/pxt-maker/issues/60

Thanks

mmoskal, 2018-07-31:
This might be mbed limitation. We have planned work to drop mbed on SAMD21 and instead use the hardware directly. That might help. Unfortunately, there is no clear timeline on that yet...

sabas1080, 2018-08-01:
Can change directly on the mbed code?

https://github.com/ARMmbed/mbed-os/blob/1f94ede86ce13b4e5bfa1a60b167ce42896a866c/targets/TARGET_Atmel/TARGET_SAM_CortexM0P/spi_api.c

sabas1080, 2018-08-01:
I opened a pull request #62

closed 2018-10-18 by sabas1080