microsoft/pxt-maker #64: Blocks serial available and read

sabas1080, 2018-08-10:
Hi @mmoskal 

My intention is to read the information of a device via serial, but I have noticed that the "Serial" package does not have the functions `Serial.available and Serial.read`

Is there any intention to add them?

Thanks

pelikhan, 2018-10-18:
Can you send a PR?

sabas1080, 2018-10-30:
Yes, but you can not test because I'm in samd21

pelikhan, 2018-10-30:
We typically HF2/WebUSB for logging. What are you testing?

sabas1080, 2018-10-30:
Communication with an module serial as GPS

pelikhan, 2018-11-05:
Try again, serial library now available.

sabas1080, 2018-11-05:
not working serial, my code is:

```
serial.setBaudRate(BaudRate.BaudRate115200)
serial.redirect(pins.TX, pins.RX)
forever(function () {
    pins.LED.digitalWrite(true)
    pause(500)
    pins.LED.digitalWrite(false)
    pause(500)
    console.logValue("x", 0)
})
```

pelikhan, 2018-11-05:
try ``serial.writeValue("x", 0)``

pelikhan, 2018-11-05:
Add
```
serial.attachToConsole()
```

sabas1080, 2018-11-05:
the intermediate blink does not work when serial insert

```
serial.setBaudRate(BaudRate.BaudRate115200)
serial.redirect(pins.TX, pins.RX)
serial.attachToConsole()
forever(function () {
    serial.writeValue("x", 0)
    pins.LED.digitalWrite(true)
    pause(500)
    pins.LED.digitalWrite(false)
    pause(500)
})
```

pelikhan, 2018-11-05:
Do you have serial working at the CODAL level?

sabas1080, 2018-11-05:
How can I do that?

pelikhan, 2018-11-05:
Write a C++ tests that uses your codal target and serial to make sure everything works at that level.

pelikhan, 2018-11-06:
> I was trying on a sparkfun samd21, it seems it does not break with the changes, tomorrow I'll see in an oscilloscope serial communication
>
> #84<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FMicrosoft%2Fpxt-maker%2Fpull%2F84&data=02%7C01%7Cjhalleux%40microsoft.com%7C590c71ba99f74aab589908d6438f8350%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636770681007066377&sdata=tru66DRE%2FLaerE6Pf4oTm2ZLB8F4NFVDPIAD1MTRgZ4%3D&reserved=0>

I will try with a metro tomorrow.

sabas1080, 2018-11-06:
Serial Working, Now I'm going to read
[image]

pelikhan, 2018-11-06:
Looks like this done.

closed 2018-11-06 by pelikhan