microsoft/pxt-maker #65: missing targetconfig.json on pxt serve

BartNetJS, 2018-08-10:
**Describe the bug**
On pxt serve the browser opens on the hosted url, but pxt serve craches with an exeception that "targetconfig.json" is missing

The same for microbit works fine. i can also see the targetconfig.json in the folder.

**To Reproduce**
Steps to reproduce the behavior:
1. md maker (from a console prompt)
2. cd maker
3. pxt target maker (no errors)
4. pxt serve
5. Error: ENOENT: no such file or directory, open 'D:\source\FC\playground\maker\node_modules\pxt-maker\targetconfig.json'

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser chrome
 - Version 67.0.3396.99

pelikhan, 2018-10-05:
The npm package is updated. Try
```
pxt update
``` 
and build again

pelikhan, 2018-11-05:
Looks like this happens when we are in the middle of building packages (and our CLI has messed up the current directoy).

pelikhan, 2018-11-23:
Use ``pxt serve --rebundle`` which is somewhat less prone to this bug.