microsoft/pxt-maker #73: Error Compile with codal MKR1300

sabas1080, 2018-10-19:
Hi @pelikhan 

 I am trying to compile a version with [codal mkr1300](https://github.com/ElectronicCats/codal-mkr1300), when trying to do it, the binary is generated, but the execution of pxt serve does not end

My log
 https://makecode.com/compile/497c9a0073cb7220b49cb8fb6a541bd0fd2d4956c1c3eb93d7dbbe349c70067e.log

pxt-core: 4.1.44
pxt commons 0.24.6

Thanks

pelikhan, 2018-10-19:
Try this to get a local build output
```
pxt serve --local
```

sabas1080, 2018-10-19:
```
***
*** Build failed: target build failed: ENOENT: no such file or directory, open 'built/codal/build/CIRCUIT_PLAYGROUND.hex'
Error: ENOENT: no such file or directory, open 'built/codal/build/CIRCUIT_PLAYGROUND.hex'
    at Object.fs.openSync (fs.js:646:18)
    at Object.fs.readFileSync (fs.js:551:33)
    at Object.patchCodalHexInfo [as patchHexInfo] (/Users/sabas/Documents/proyectoshd/pxt-produccion/pxt/built/buildengine.js:120:17)
    at /Users/sabas/Documents/proyectoshd/pxt-produccion/pxt/built/pxt.js:133248:56
    at tryCatcher (/Users/sabas/Documents/proyectoshd/pxt-produccion/pxt/node_modules/bluebird/js/release/util.js:16:23)
    at Promise._settlePromiseFromHandler (/Users/sabas/Documents/proyectoshd/pxt-produccion/pxt/node_modules/bluebird/js/release/promise.js:512:31)
    at Promise._settlePromise (/Users/sabas/Documents/proyectoshd/pxt-produccion/pxt/node_modules/bluebird/js/release/promise.js:569:18)
    at Promise._settlePromise0 (/Users/sabas/Documents/proyectoshd/pxt-produccion/pxt/node_modules/bluebird/js/release/promise.js:614:10)
    at Promise._settlePromises (/Users/sabas/Documents/proyectoshd/pxt-produccion/pxt/node_modules/bluebird/js/release/promise.js:693:18)
    at Async._drainQueue (/Users/sabas/Documents/proyectoshd/pxt-produccion/pxt/node_modules/bluebird/js/release/async.js:133:16)
    at Async._drainQueues (/Users/sabas/Documents/proyectoshd/pxt-produccion/pxt/node_modules/bluebird/js/release/async.js:143:10)
    at Immediate.Async.drainQueues (/Users/sabas/Documents/proyectoshd/pxt-produccion/pxt/node_modules/bluebird/js/release/async.js:17:14)
    at runCallback (timers.js:789:20)
    at tryOnImmediate (timers.js:751:5)
    at processImmediate [as _immediateCallback] (timers.js:722:5)
***
```

pelikhan, 2018-10-19:
Try
```
pxt clean
```
and build again.

sabas1080, 2018-10-19:
apparently the folder `built/codal/build/file.hex`
 is not being generated

pelikhan, 2018-10-19:
If you are building for a new codal, you are going to need a new variant in pxttarget.json and a new build service. We will need to setup the build service.

sabas1080, 2018-10-19:
What is the easiest way to do this? everything arises as a result of this https://github.com/lancaster-university/mbed-classic/pull/10

pelikhan, 2018-10-19:
We'll setup the build tomorrow.

pelikhan, 2018-10-19:
This is for the arduino-mkr-1300 board right?

wero1414, 2018-10-19:
Hi @pelikhan yes is for the MKR-1300

pelikhan, 2018-10-19:
Btw we have a discord to speed up comm https://aka.ms/makecodecommunity

pelikhan, 2018-10-19:
See PR https://github.com/Microsoft/pxt-maker/pull/75

sabas1080, 2018-10-20:
thanks @pelikhan , but when trying to compile, I have the next error:

`{ Error: Bad HTTP status code: 412 at https://www.makecode.com/api/compile/extension; message: {"message":""}`

pelikhan, 2018-10-20:
In master?

sabas1080, 2018-10-20:
I reinstalled dependencies and everything ok, thanks @pelikhan

closed 2018-10-20 by sabas1080