microsoft/pxt-maker #98: Serial not working with Arduino Zero

louisvangeldrop, 2018-11-22:
**Describe the bug**

As sson as I use a serial function such as serial.redirect(pins.TX, pins.RX, BaudRate.BaudRate115200) the arduino zero stops working.

A clear and concise description of what the bug is.

**To Reproduce**

let pir = false
serial.redirect(pins.TX, pins.RX, BaudRate.BaudRate115200)
loops.forever(function () {
    pause(1000)
    pir = !pir //pins.D7.digitalRead()
    if (pir) {
        pins.RXLED.digitalWrite(true)
        pins.LED.digitalWrite(true)
    } else {
        pins.RXLED.digitalWrite(false)
        pins.LED.digitalWrite(false)
    }
})

The LED's are not flashing

When I comment out the serial.redirect, it works

sabas1080, 2018-11-22:
Hi @louisvangeldrop 

try with the solution of the issue https://github.com/Microsoft/pxt-maker/issues/64

louisvangeldrop, 2018-11-22:
I can't find a solution in issue  #64. Maybe I missed something

sabas1080, 2018-11-22:
What version of pxt-maker are you using? have you verified that the pins of TX and RX are correctly declared in the config.ts of the arduino zero?

louisvangeldrop, 2018-11-23:
Local version installed 2 days ago. I have the same errro using the maker website maker.makecode.com
 The pins on the board are: RX =D0 and TX=D1. However in config.ts de spec is:
 export const PIN_RX = DAL.PB03;
 export const PIN_TX = DAL.PA27;
export const PIN_D0 = DAL.PA11;
 export const PIN_D1 = DAL.PA10;

louisvangeldrop, 2018-11-23:
I have modified config.ts  by setting RX to PA11 and TX to PA10 and it works now

louisvangeldrop, 2018-11-23:
No idea how to do that, but never too old ( 72 years) to learn how to do that.
PS. I have just read that also the Arduino Uno is supported. Couldn't find the target library

pelikhan, 2018-11-23:
Unfortunately the Uno (AVR) is not supported.

pelikhan, 2018-11-23:
Could you add your modified code here?

Here is a guide to PRs. It is a bit mind boggling at a start but GitHub makes it easier.
https://help.github.com/articles/creating-a-pull-request-from-a-fork/

louisvangeldrop, 2018-11-23:
The article about Makecode and Codal gave me some hope.

Here is the modified config.ts

namespace config {
  export const NUM_NEOPIXELS = 0;

  export const PIN_A0 = DAL.PA02;
  export const PIN_A1 = DAL.PB08;
  export const PIN_A2 = DAL.PB09;
  export const PIN_A3 = DAL.PA04;
  export const PIN_A4 = DAL.PA05;
  export const PIN_A5 = DAL.PB02;

  export const PIN_D0 = DAL.PA11;
  export const PIN_D1 = DAL.PA10;
  export const PIN_D2 = DAL.PA14;
  export const PIN_D3 = DAL.PA09;
  export const PIN_D4 = DAL.PA08;
  export const PIN_D5 = DAL.PA15;
  export const PIN_D6 = DAL.PA20;
  export const PIN_D7 = DAL.PA21;
  export const PIN_D8 = DAL.PA06;
  export const PIN_D9 = DAL.PA07;
  export const PIN_D10 = DAL.PA18;
  export const PIN_D11 = DAL.PA16;
  export const PIN_D12 = DAL.PA19;
  export const PIN_D13 = DAL.PA17;
  export const PIN_LED = DAL.PA17;
  export const PIN_RX = DAL.PA11; //DAL.PB03;
  export const PIN_TX = DAL.PA10; //DAL.PA27;
  export const PIN_MISO = DAL.PA12;
  export const PIN_MOSI = DAL.PB10;
  export const PIN_SCK = DAL.PB11;
  export const PIN_SCL = DAL.PA23;
  export const PIN_SDA = DAL.PA22;
  export const PIN_RXLED = DAL.PB03;
  export const PIN_TXLED = DAL.PA27;
  export const PIN_ACCELEROMETER_INT = DAL.PA10; // D1
  export const PIN_ACCELEROMETER_SCL = DAL.PA23; // SCL
  export const PIN_ACCELEROMETER_SDA = DAL.PA22; // SDA
}

/*
// for M0
namespace userconfig {
    export const PIN_D2 = DAL.PA08;
    export const PIN_D4 = DAL.PA14;
}
*/

pelikhan, 2018-11-23:
Which pinout did you use? The schematics on Arduino.org?

louisvangeldrop, 2018-11-25:
I used the pins TX=D1 and RX =D0 as shown in the makecode editor picture