microsoft/pxt-maker #147: Support for Wemos D1 mini / ESP8266

ivanporro, 2019-01-19:
**Is your feature request related to a problem? Please describe.**
It seems to me the system is missing support for ESP8266 boards. Is anyone already working on this? is it planned?

**Describe the solution you'd like**
I'm willing to add support by myself if someone can point out to method / issues and confirm it is feasible.

**Describe alternatives you've considered**
None at the moment.

**Additional context**
SVG part for the Wemos D1 mini:
https://github.com/mcauser/Fritzing-Part-WeMos-D1-Mini

pelikhan, 2019-01-23:
The ESP8266 runs a chipset our compiler does not support. It is unlikely we will add support for that chip in the near future. We do have plans to communicate with the WiFi module but not reprogram it.

ivanporro, 2019-01-24:
Thank you very much, pretty clear.

closed 2019-01-24 by ivanporro