microsoft/pxt-maker #149: WebUSB or console device

sabas1080, 2019-01-23:
**Describe the bug**
In version beta

MakeCode Maker (Cortex class) version:  0.6.4
Microsoft MakeCode version:  5.4.7

webusb only work when device reset in mode boot and download firmware. previously pair from the menu configuration

**To Reproduce**
Steps to reproduce the behavior:
1. Pair device
2. Download firmware with console.log include and never ends
3. Reset device with double boton click, enter in mode boot and download firmware
4. See error

**Desktop (please complete the following information):**
 - OS: MacOS
 - Browser chrome
 - Version 71

**Additional context**
tested in mrk1300 and meow meow

pelikhan, 2019-01-23:
Could add the code snippet you are using?

sabas1080, 2019-01-23:
ready

pelikhan, 2019-02-05:
Console fixed.

closed 2019-02-05 by pelikhan