microsoft/pxt-maker #187: RedBoard Turbo w/ Music Blocks

bboyho, 2019-02-25:
**Describe the bug**
Issue with "Music" blocks. The buzzer does not seem to be functioning on A0 when using it with the SparkFun RedBoard Turbo under MakeCode Beta site.

**To Reproduce**
When attaching a piezo buzzer to the DAC on A0, it does not seem to work. "Adafruit Circuit Playground Express" seems to work on pin A0. 
1. Drag the Music > play sound "ba ding"code block.
2. Insert into the forever loop.

**Expected behavior**
The board should make a sound when the code executes the forever loop. Specifically a coin sound effect.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 (64-bit)
 - Browser: Chrome
 - Version:  72.0.3626.109 (Official Build) (64-bit)

pelikhan, 2019-02-25:
Use “play sound until done” as play sound is non-blockjng.

bboyho, 2019-02-25:
I tried using the "play sound ba ding until done" code block but there is no sound coming out. Adding a pause after did not help either. :'( 

[image]

pelikhan, 2019-02-26:
There is an amplifier on the CPX so you get to hear. I've changed the image to a headphone so that it actually gets to be hearable.

bboyho, 2019-02-28:
Ah, I did not realize that until you pointed it out. I thought it was a piezo that was on the CPX. I used the same code but I was not able to get it working on a piezo buzzer. Is it possible to set it to output a PWM signal to control a piezo? Or would I need to get an audio amplifier and a small speaker? I feel like it would be easier for users if they use it with the piezo buzzer.

pelikhan, 2019-03-04:
There is no amp so we would need to emit a square wave at full volume to get something out.