microsoft/pxt-maker #188: RedBoard Turbo w/ External Addressable LEDs

bboyho, 2019-02-25:
**Describe the bug**
Adding an external WS2812-based LEDs from the "Light" blocks seems to not fully work on other pins when using it with the SparkFun RedBoard Turbo under MakeCode Beta site. There seems to be issues controlling LEDs after 3 or 4 pixels. There also seems to be issues setting up all the LEDs to the same color. The on-board WS2812B stays lit with the color green even when not calling it.

Maybe I am setting it up incorrectly?

**To Reproduce**
1. Connect external WS2812s to A1, 3.3V, and GND.
2. Drag the `Light > More > set strip to create WS2812 strip on LED with 30 pixels` code block
3. Change `LED` to `A1`.
4. Change `30` pixels to `5` pixels.
5. Drag `Light > More > set strip brightness 20`.
6. Change brightness `20` to `150`.
7. Drag the `Light > More > set strip show animation for 2000 ms` code block.
8. Drag the `Light > More > set strip all pixels to blue` code block.
9. Add a few pauses between the animation and setting all LEDs to the same color.

**Expected behavior**
The 5x external WS2812 LEDs should animate and then all change to the color blue.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: * Windows 10 (64-bit)
 - Browser: 
 - Version: * 72.0.3626.109 (Official Build) (64-bit)

pelikhan, 2019-02-26:
@bboyho you can click on the "share" button and copy the URL generated as well. Or copy paste the image generated by the "save" button.

pelikhan, 2019-02-26:
The following programs works locally both for A1 and MOSI
```
let on = false;
forever(function () {
    pins.LED.digitalWrite(on = !on)
    pause(100)
});

let strip = light.createNeoPixelStrip(pins.A1, 60)
forever(function () {
    strip.showAnimation(light.rainbowAnimation, 2000)
    strip.setAll(0xff0000)
    pause(500)
    strip.setAll(0x00ff00)
    pause(500)
    strip.setAll(0x0000ff)
    pause(500)
})
light.setLength(60)
forever(function () {
    light.showAnimation(light.rainbowAnimation, 2000)
    light.setAll(0x0000ff)
    pause(2000)
})
```

pelikhan, 2019-02-28:
Reopen if you have a hardware repro. Video helps.

closed 2019-02-28 by pelikhan

bboyho, 2019-02-28:
As a quick note, I tested this with a second set of WS2812 LEDs on a strip and it works. The initial test must have had a bad set of addressable LEDs.  False alarm.

Thanks for your patience. =)

pelikhan, 2019-02-28:
I also tested a strip of APA102s on the lumidrive — all good. If you mount them on the SPI pin, you’ll get DMA accelerated as well.