microsoft/pxt-maker #193: RedBoard Turbo w/ Buttons

bboyho, 2019-02-28:
**Describe the bug**
There seems to be no internal pull-up resistor for a button press leaving any button floating as shown in the simulator.  I'm not sure if we would need to include an image of a pull-up resistor in the simulator just like the LED's current limiting resistor for those that happen to use a button/switch?

**To Reproduce**
When attaching a momentary push button on any I/O pin, the pin is floating.
1. Drag the `logic > if then` code block in to the forever.
2. Drag and set `button D8 is pressed` into the if's condition statement.
3. Drag and set `pin > digital write pin LED to HIGH` as the result.
4. Drag and set `pin > digital write pin LED to LOW` under the else's result.

This can be fixed with a physical pull-up resistor to the 3.3V pin or setting the pin as an internal pull-up code block (i.e. `set pull pin ____ to up`).  That might be more work than necessary since it depends on the setup. It will probably just need an example for documentation instead of having it in the simulator?

**Expected behavior**
Pushing a button should turn on the built-in LED on pin 13. When the button is not pressed, the LED should remain off. 

**Screenshots**
[https://makecode.com/_FiU3hK5AADCW](https://makecode.com/_FiU3hK5AADCW)

**Desktop (please complete the following information):**
 - OS: Windows 10 (64-bit)
 - Browser: Chrome
 - Version:  72.0.3626.109 (Official Build) (64-bit)

pelikhan, 2019-02-28:
This "set pull pin up" happens implicitely when using the pin as a button.

In https://github.com/Microsoft/pxt-maker/blob/master/libs/sparkfun-samd21-dev-breakout/device.d.ts#L181, the button is configured as active low / pull up. And in the button constructor (https://github.com/lancaster-university/codal-core/blob/master/source/drivers/Button.cpp#L55 ), the pull is set on the pin.

Are you observing an incorrect behavior on hardware?

bboyho, 2019-02-28:
Yeah, I am observing a different behavior. If the button is not being pressed down, the pin is floating. It just needs an internal pullup resistor in the setup function =>[https://makecode.com/_d9L8hHDdHTHJ](https://makecode.com/_d9L8hHDdHTHJ) .

The wiring diagram in the simulator seemed misleading if the internal pullup resistor was not included.  Seems more like a minor detail.

pelikhan, 2019-03-01:
Ok i see now the confusion. When using "button D2 is pressed", we automatically set the pull up on the pin. However we use pin digital read, we don't set any pull. The simulator should reflect that.

pelikhan, 2019-03-01:
So "button D8 is pressed" set the pull up, "pin D8 digital read" does not set any pull.

bboyho, 2019-03-01:
To clarify, the first example using D8 did not have an internal pull-up resistor in the setup so the button read was floating. The wiring diagram in the simulator seemed misleading if the internal pullup resistor was not included. 

The second example using D2 utilized the internal pull-up resistor in the setup so the button read worked as expected. The simulator was fine since we were using an internal pull-up resistor.

pelikhan, 2019-03-01:
“button D8 is pressed” does set the pull up on D8 (it should at least. I’ll check with a scope.