microsoft/pxt-maker #197: RedBoard Turbo w/ Servo Block

bboyho, 2019-03-01:
**Describe the bug**
Issue with the servo blocks. Attaching a servo on the analog pins does not seem to work when using it with the SparkFun RedBoard Turbo under the MakeCode Beta site.

I am not sure if I am doing it correctly?

**To Reproduce**
Servo does not seem to move when it is connected to the analog pins A0-A5.

1. Drag and set `pins > servo write pin A1 to 0` in the forever loop.
2. Drag and set `loops > pause 1000 ms`.
3. Drag and set `pins > servo write pin A1 to 180`.
4. Drag and set  `loops > pause 1000 ms`.

**Expected behavior**
The servo should move between two positions every second on a PWM pin. The example works when the servo is connected to the PWM pins between D3-D13 but it does not work on A0-A5. Looking at the pinout, it does not seem like the RedBoard Turbo has any PWM pins on A0-A5?

**Screenshots**
https://makecode.com/_FPrd0969PMDA

**Desktop (please complete the following information):**
 - OS: Windows 10 (64-bit)
 - Browser: Chrome
 - Version:  72.0.3626.109 (Official Build) (64-bit)

pelikhan, 2019-03-01:
This is the pin definitions. https://github.com/Microsoft/pxt-maker/blob/master/libs/sparkfun-redboard-turbo/device.d.ts

https://makecode.com/_FPrd0969PMDA<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fmakecode.com%2F_FPrd0969PMDA&data=02%7C01%7Cjhalleux%40microsoft.com%7C0dbd10368f3e4dec728d08d69e943281%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636870756669513050&sdata=zRguxMzBQjYqVwDQKxqQjsMNi4NlBZR3CUbcG3tgvmE%3D&reserved=0>

pelikhan, 2019-03-01:
This is where pins are defined: https://github.com/Microsoft/pxt-maker/blob/master/libs/sparkfun-redboard-turbo/device.d.ts

pelikhan, 2019-03-02:
Fixed by https://github.com/Microsoft/pxt-maker/commit/9028b1974575982e1d65a9dcfb8c35f99b9bb670

closed 2019-03-02 by pelikhan