microsoft/pxt-maker #204: Issue using LCD with Adafruit Metro M4

brentru, 2019-03-08:
**Describe the bug**
I've wired an Adafruit Metro M4 the same way the Metro M4 in the editor window is wired. Then I and added the `LCD Show String` block to print out "hello world". The LCD shows nothing, though it is turned on .

**To Reproduce**
Steps to reproduce the behavior:
1. Go to LCD
2. Click on _LCD Show String "Hello World" at line 1_

**Expected behavior**
LCD should display Hello World

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser: Chrome

**Additional context**
As a side-note, the LCD should be connected to the Metro's +5V instead of +3.3V (using this standard 16x2 LCD for reference; https://www.adafruit.com/product/181)

pelikhan, 2019-03-08:
Could you try using the 5V rail for the large LCD? I need to add support for 5V in our routing algorithm.

pelikhan, 2019-03-08:
Might be the brightness pin that is incorrect.

brentru, 2019-03-08:
@pelikhan  I used the 5V rail for the 16x2 LCD when I wired it up on the breadboard. Could you paste the pinouts in here (they're a bit strange on the website, not sure if they're correct)

pelikhan, 2019-03-08:
I following this guide/pinout for the wiring: https://learn.adafruit.com/experimenters-guide-for-metro/circ14-wiring . The pinout is here: https://github.com/Microsoft/pxt-maker/blob/master/libs/adafruit-metro-m4-express/config.ts#L58

pelikhan, 2019-03-09:
I was missing the ground wire from RW
[image]

pelikhan, 2019-03-09:
@brentru please try again (v0,9.16). it routes to 5v and has the missing round wire.
[image]

pelikhan, 2019-03-11:
Fixed.

closed 2019-03-11 by pelikhan

brentru, 2019-03-11:
@pelikhan  Thanks for fixing, works on my end.