microsoft/pxt-maker #212: support for Beaglebone boards

vaishnav98, 2019-03-25:
**Is your feature request related to a problem? Please describe.**

This feature request is supposed to initiate a discussion regarding adding support for [Beaglebone boards](https://beagleboard.org/boards) in the pxt-maker target, I have been working on a separate [Makecode target](https://github.com/vaishnav98/pxt-beaglebone), starting from the pxt-maker sources to include support for the Beaglebone boards and was [successful(live simulator)]( https://vaishnav98.github.io/pxt-beaglebone/) in implementing basic support for the Beaglebone Boards.

I would like to work on integrating my solution with the pxt-maker repository and bring in Beaglebone Boards support directly through the Makecode Maker target rather than creating a separate target for the same.

**Describe the solution you'd like**

I would like to have the Beaglebone Boards support delivered directly through Makecode Maker target rather than creating a separate target for the same. I am willing to work on the integration and would like to hear your valuable feedback on the same.

**Describe alternatives you've considered**

I have tried to create a new PXT target starting from the pxt-maker sources for providing the Beaglebone Support and was successful in implementing basic support for the same. I believe that it would be better to have the Beaglebone support served through the Makecode Maker target and would like to work on integrating the existing solution with the pxt-maker codebase. 

**Additional context**

I have uploaded some quick test videos for the pxt-beaglebone target which you can find here : 

- https://www.youtube.com/watch?v=niSkdlEI5UE
- https://www.youtube.com/watch?v=n2EkLOWMtDc
- https://www.youtube.com/watch?v=5YAto5vOWnA

Please let me know if you require any other details from me, Thank you.

pelikhan, 2019-03-25:
The editor looks super cute :)

Yes we love to see beaglebone support in maker.makecode.com . The best way to go forward is to prepare pull requests for the required changes. At first, you can mark the beaglebone boards as "- beta" until the integration is tested. We are also on discord aka.ms/makecodecommunity.

Thank you for your work!

pelikhan, 2019-03-25:
Do you have the end-to-end compilation figured out?

vaishnav98, 2019-03-28:
Thank you :) @pelikhan , I'll prepare pull requests for the Beaglebone support in the Makecode Maker target, the compilation part is something I am a bit confused with, sure I'll join the Discord community and continue further discussions there :)

closed 2019-03-28 by vaishnav98