microsoft/pxt-maker #217: Download Error when running Local Dev Server

JoeBakalor, 2019-04-16:
**Describe the bug**
Error when trying to download code when running Local Dev Server but not on the website. 

**To Reproduce**
Steps to reproduce the behavior:
1. Run "pxt serve" to start local dev server
2. Click on "New Project"
3. Select any board
4. Select "Download"

It looks like it always tries to download to the device instead of downloading the hex file when running a local instance 

**Expected behavior**
The expectation is for the file to be downloaded and a popup window to show "Download Completed" 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS Mojave Version 10.14.3
 - Browser: Chrome
 - Version: 73.0.3683.103

**Additional context**
Maybe there is a configuration that needs to be set for running locally that I am not doing?  I did not have this issue running a local instance in January.  I even tried checking out an earlier commit when I know it was working and I still get the same error.

smith-matt, 2019-05-07:
I wanted to bump this. I'm receiving the same error with any board. This is hindering any further development. Is anyone at Microsoft/ @pelikhan able to reproduce this?

pelikhan, 2019-05-07:
Looks like "hid" is installed and we fail to deploy.

pelikhan, 2019-05-07:
or “WebUSB”. Fix coming.

pelikhan, 2019-05-08:
The issue should be fixed for you on the latest build. Make sure to run npm install again. Reopen if the issue is still there.

closed 2019-05-08 by pelikhan

smith-matt, 2019-05-11:
@pelikhan I'd like to re-open this issue.

**Describe the bug**
Compile issue/error when trying to download code when running Local Dev Server but not on the website.

**To Reproduce**
Steps to reproduce the behavior:

1. Run "pxt serve" to start local dev server
2. Click on "New Project"
3. Select any board
4. Select "Download"

It looks like it's still trying to download to the device, instead of allowing the download of the .hex or .uf2 file. 

**Expected behavior**
The expectation is for the file to be downloaded and a popup window to show "Download Completed"

**Screenshots**
There are two cases:

1. Project doesn't create a .uf2 file (BOARD = teknikio-bluebird)

[image]

2. Project does create a .uf2 file (BOARD = Adafruit Metro M0 Express)

[image]

**Desktop (please complete the following information):**
 - OS: macOS Mojave Version 10.14.4
 - Browser: Chrome
 - Version: Version 73.0.3683.103 (Official Build) (64-bit)

**Additional context**
I would like to be able to download a .hex file from the teknikio-bluebird project. This will allow us to continue development on the local dev server so that we may tweak and test the build process. It looks like it is almost there, but the offline dev server doesn't behave the same way as the online/web version. If any of this is unclear please let me know and I can explain more.

JoeBakalor, 2019-05-13:
@pelikhan I just updated to the latest build and I am seeing the same problem as @smith-matt