microsoft/pxt-maker #229: Re-opening: Download Error when running Local Dev Server #217

JoeBakalor, 2019-05-14:
@pelikhan I'd like to re-open this issue.

**Describe the bug**
Compile issue/error when trying to download code when running Local Dev Server but not on the website.

**To Reproduce**
Steps to reproduce the behavior:
1. Run "pxt serve" to start local dev server
2. Click on "New Project"
3. Select any board
4. Select "Download"

It looks like it's still trying to download to the device, instead of allowing the download of the .hex or .uf2 file.

**Expected behavior**
The expectation is for the file to be downloaded and a popup window to show "Download Completed"

**Screenshots**
There are two cases:

1. Project doesn't create a .uf2 file (BOARD = teknikio-bluebird)
[image]

2. Project does create a .uf2 file (BOARD = Adafruit Metro M0 Express)
[image]

**Desktop (please complete the following information):**
- OS: macOS Mojave Version 10.14.4
- Browser: Chrome
- Version: Version 73.0.3683.103 (Official Build) (64-bit)

**Additional context**
I would like to be able to download a .hex file from the teknikio-bluebird project. This will allow us to continue development on the local dev server so that we may tweak and test the build process. It looks like it is almost there, but the offline dev server doesn't behave the same way as the online/web version. If any of this is unclear please let me know and I can explain more.

pelikhan, 2019-05-14:
let me try this tonight.

pelikhan, 2019-05-16:
Ok i can reproduce the issue for bluebird

pelikhan, 2019-05-16:
It is fixed for me now. Please try out the latest build.

closed 2019-05-16 by pelikhan