microsoft/pxt-maker #255: Create strip error

zostay, 2019-10-05:
**Describe the bug**
Getting a typescript error when creating a WS8201 strip.

**To Reproduce**
Just use the sample code for "create script":

```typescript
let strip = light.createStrip(pins.A1, 10);
strip.setAll(0x00ff00)
```

**Expected behavior**
No error when using create strip.

**Desktop:**
 - OS: macOS Mojave 10.14.5
 - Browser: Chrome
- Version:  Version 78.0.3904.44 (Official Build) beta (64-bit)

**Additional context**
I have not found that this error depends on any of the parameters. The error is always `Cannot read property 'kind' of undefined`. No error using the other `.create*` constructors.

zostay, 2019-10-05:
Scratch that. I hadn't test `.createNeoPixelStrip`. It gets the same error. It's just `.createAPA102Strip` that works with no error. Too bad I don't have any of those.

zostay, 2019-10-05:
Oh, and this is happening while working on a Gemma M0 on maker.makecode.com

zostay, 2019-10-05:
Aha, I found a fix. It looks like the problem by digging around in the code. The 3rd and undocumented parameter to `.createNeoPixelStrip` needs to be set. Setting it to `NeoPixelMode.RGB` fixes the problem.

So, the bug is twofold:

1. The documentation is wrong.
2. The block should include the 3rd argument or the 3rd argument should have the default value applied correctly.

chemojay, 2019-10-08:
Thanks for your info I was able to fix my project because of you.

pelikhan, 2019-10-08:
This is a compiler bug. Adding the missing the NeoPixelMode argument fixes the issue.
```
let strip = light.createStrip(pins.A1, 10, NeoPixelMode.RGB);
strip.setAll(0x00ff00);
```

pelikhan, 2019-10-08:
looking into this

pelikhan, 2019-10-12:
Fixed.

closed 2019-10-12 by pelikhan