microsoft/pxt-maker #265: Python indention issues

gbaman, 2019-11-21:
**Describe the bug**
When writing code in the Python editor, 2 related issues.

1. When hitting enter after a colon, would expect the following section to be indented.
2. When switching back and forth (see gif) between python/blocks/python, some odd indention can occur...

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new program with a forever loop and a block with in.
2. Switch to Python view.
3. Select the colon on the function and hit enter.

**Expected behavior**
When the enter key is pressed on a line ending in a colon, it should auto indent the next line.

When converting between blocks and Python, would expect blank lines to be kept and double indenting not to be used?

**Screenshots**
Auto indent on colon issue
[image]

Python -> Blocks -> Python indentation issue
[image]

**Desktop (please complete the following information):**
 - OS: Mac Os
 - Browser: Firefox
 - Version: 70

darzu, 2019-11-22:
Thanks for the great repro.

darzu, 2019-11-22:
We're working on a bunch of Python fixes and improvements as a top priority over the next few months.

pelikhan, 2019-12-02:
A bunch of fixes are live now.

darzu, 2019-12-02:
Partially fixed by: https://github.com/microsoft/pxt/pull/6270
not yet integrated into maker; more fixes coming

darzu, 2019-12-04:
I duped the remaining work issue to: https://github.com/microsoft/pxt-minecraft/issues/1471
since we're first releasing python in Minecraft so I'm tracking key issues there.

pelikhan, 2020-01-22:
Fixed.

closed 2020-01-22 by pelikhan