microsoft/pxt-maker #275: error TS9668

wallarug, 2020-01-21:
Hi there!

Been following along with the make a board instructions and hitting some road blocks.  When I try to compile my custom board errors are thrown for all non-standard named pins.

```
config.ts(27,5): error TS9268: can't find DAL.CFG_PIN_SERVO7
```

I have been struggling to sort this one out for a while now.   Looking for ideas why, where and what is causing this issue.

I can see that config.ts has the issue but I do not understand why `DAL.CFG` is appearing in front of the names.

Any assistance would be great.

wallarug, 2020-01-21:
I have done a further investigation and discovered the `dal.d.ts` file.

There is limited documentation on how to add to this / what is the correct procedure.

pelikhan, 2020-01-21:
Can you share a repo or a pr?

wallarug, 2020-01-21:
Sure thing.

http://GitHub.com/wallarug/pxt-maker

The file I mentioned before says it is autogenerated.

wallarug, 2020-01-22:
Hey @pelikhan 

Please see this [commit ](https://github.com/wallarug/pxt-maker/commit/4af6458c931868403bdedf176af1d565d08b9e67) for what we did to make it work.

I don't believe this is the official way to fix this issue - there must be a documented, more structured way to get it working.

Thanks

@wallarug

pelikhan, 2020-01-22:
Please merge the latest changes so that all the svg stuff goes away.

wallarug, 2020-01-23:
> Please merge the latest changes so that all the svg stuff goes away.

Hey @pelikhan - cool I'll do that.

That does not fix this issue though.  I still cannot use custom pin names without hacking up some files that say 'please do not change'.  I think it is related to the **dependencies** in `libs/<board-name>/pxt.json`.

These **dependencies** seem to be sourced from: `pxt-maker/node_modules/pxt-common-packages/libs/` and `pxt-maker/libs/core---samd51/dal.d.ts`

Is there a way we can add our pin names (SERVO1, SERVO2, etc) into a file for support?

pelikhan, 2020-01-23:
You need to add entries in configkeys.h in the Microsoft/pxt-common-packages, rebuild the dal.d.ts files, then you will be able to use those in your project. It’s a bit tricky to get right.

What are the keys you need? You cannot use the A0..A32, B0..32, P0..32 instead?

wallarug, 2020-01-24:
We can, but would like to use what is printed on the board.  If Adafruit and other companies can use their custom pin names we would like to be able to (or at least know how to) as well.

I'll have a look for the `configkeys.h` file in that repository.

pelikhan, 2020-01-24:
Try runnning 

   pxt builddaldts

From maker after you’ve updated that file. Make sure pxt-common-packages is linked as mentionned in the readme.

wallarug, 2020-01-24:
Thanks!

I'm trying this now.  I've developed a few other issues along the way that I also need to sort out.

Out of interest, what do you use to develop the SVG images?

pelikhan, 2020-01-24:
InkScape. Might want to try “pxt clean” first.

peterpanstechland, 2020-02-04:
hi @pelikhan I am working together with @wallarug on porting MakeCode to our Robo HAT MM1 board, I think I have successfully modified the configkey.h, but when i do `pxt buidaldts` in the pxt-maker folder, I only get `core---samd/dal.d.ts` updated but not `core---samd51/dal.d.ts`.

pelikhan, 2020-02-04:
I ran
```
pxt builddaldts
```
and got all files to update. see https://github.com/microsoft/pxt-maker/pull/280/files

pelikhan, 2020-02-04:
Yes please open a PR in common packages.

pelikhan, 2020-02-14:
I'm going to close this. See PR on more docs for building CODAL --> https://github.com/microsoft/pxt-maker/pull/289

closed 2020-02-14 by pelikhan

wallarug, 2020-02-15:
Thanks @pelikhan !