microsoft/pxt-maker #279: Is there a way to use a private repository for codalTarget?

OpusK, 2020-02-04:
Hi, 

Can I use a private repository with codalTarget's git repository?
In [yotta, "git + ssh"](http://docs.yottabuild.org/tutorial/privaterepos.html) seems to play the role, but is there a way to do it in codal?

```json
                "codalTarget": {
                    "name": "codal-nrf52840-dk",
                    "url": "https://github.com/mmoskal/codal-nrf52840-dk",
                    "branch": "v1.1.0",
                    "type": "git"
                },
```

pelikhan, 2020-02-04:
Is this to build locally or to build in our cloud?

OpusK, 2020-02-04:
Do the settings differ depending on the build location?
I'm currently building locally.

pelikhan, 2020-02-04:
Using a terminal with git credentials should work. Cloud build of private repoes require authorization on our part

OpusK, 2020-02-05:
Thanks, @pelikhan :)
I understood.

closed 2020-02-05 by OpusK