microsoft/pxt-maker #282: Docs Update - Custom Boards

wallarug, 2020-02-05:
Hi there,

I'll take ownership of this one and formalise it a bit better.

This is relating to adding custom boards with extended name sets.

@peterpanstechland message:
in order to get the dts.ts.d autogenerated, you need to(this is what i did):
```
pxt clean
rm -Rf node_modules/pxt-common-packages
pxt link ../pxt-common-packages  (this is after you have modified the configkeys.h)
pxt buidaldts (if you get some error about missing file in the built folder, I got this error when i did this, and then I tried run "pxt builddaldts" in the pxt-maker/node_modules/pxt-common-package/ first, then run this command again in the pxt-maker folder)
this should update the dts.ts.d files.
```

pelikhan, 2020-02-14:
I've added instructions in the README on building CODAL: https://github.com/microsoft/pxt-maker/pull/289

pelikhan, 2020-02-14:
I've added a note about dal.d.ts too