microsoft/pxt-maker #308: Server problem for deploy.

Kunal614, 2020-04-20:
I changes package.json start="pxt serve "
Then i deployed in google vm but shows 502 bad gatewaye

{
    "name": "pxt-core",
    "version": "5.37.26",
    "description": "Microsoft MakeCode, also known as Programming Experience Toolkit (PXT), provides Blocks / JavaScript tools and editors",
    "keywords": [
        "TypeScript",
        "JavaScript",
        "Blockly",
        "education",
        "microbit",
        "Adafruit",
        "pxt",
        "Microsoft",
        "Microsoft MakeCode"
    ],
    "repository": {
        "type": "git",
        "url": "git+https://github.com/microsoft/pxt.git"
    },
    "author": "Microsoft Corporation",
    "license": "MIT",
    "homepage": "https://github.com/microsoft/pxt#readme",
    "files": [
        "README.md",
        "built/*.js",
        "built/*.d.ts",
        "built/pxt-common.json",
        "built/web/fonts",
        "built/web/*.js",
        "built/web/*.css",
        "built/web/*.html",
        "built/web/vs/editor/editor.main.js",
        "built/web/vs/language/typescript/lib/*.js",
        "built/web/vs/language/typescript/src/*.js",
        "pxtcompiler/ext-typescript/lib/lib.d.ts",
        "pxtcompiler/ext-typescript/lib/typescript.js",
        "pxtcompiler/ext-typescript/lib/typescriptServices.d.ts",
        "libs/lang-test0/*.ts",
        "libs/lang-test0/pxt.json",
        "pxtarget.json",
        "webapp/public",
        "typings/**",
        "localtypings/**",
        "tests/blocks-test/karma.conf.js",
        "built/tests/blocksrunner.js",
        "docfiles",
        "theme",
        "common-docs"
    ],
    "main": "built/pxt.js",
    "engines": {
        "node": ">=8.0.0",
        "npm": ">=3.0.0"
    },
    "dependencies": {
        "applicationinsights-js": "^1.0.20",
        "bluebird": "3.5.1",
        "browserify": "16.2.0",
        "chai": "^3.5.0",
        "cssnano": "4.1.10",
        "dompurify": "2.0.8",
        "faye-websocket": "0.11.1",
        "karma": "^5.0.2",
        "karma-chai": "0.1.0",
        "karma-chrome-launcher": "3.1.0",
        "karma-mocha": "^2.0.0",
        "less": "2.7.3",
        "lzma": "2.3.2",
        "marked": "^0.8.2",
        "mocha": "5.1.0",
        "pngjs": "3.4.0",
        "postcss": "6.0.21",
        "puppeteer": "^2.0.0",
        "pxt": "^0.5.1",
        "request": "2.88.0",
        "rimraf": "2.5.4",
        "rtlcss": "2.2.1",
        "sanitize-html": "1.22.0",
        "semantic-ui-less": "2.2.14",
        "uglify-js": "3.3.21",

    },
    "lazyDependencies": {
        "node-hid": "^0.7.2",
        "webusb": "^1.1.1"
    },
    "devDependencies": {
        "monaco-editor": "0.19.3",
        "pxt-blockly": "3.0.6",
        "react": "16.8.3",
        "react-dom": "16.11.0",
        "react-modal": "3.3.2",
        "react-redux": "5.1.1",
        "react-tooltip": "3.9.0",
        "redux": "3.7.2",
        "pouchdb": "7.2.1",
        "pouchdb-adapter-memory": "7.2.1",
        "typescript": "3.7.5",
        "jquery": "3.4.1",
        "fuse.js": "3.2.0",
        "highlight.js": "9.12.0",
        "smoothie": "1.35.0",
        "envify": "4.1.0",
        "uglifyify": "5.0.2",
        "tslint": "^5.9.1",
        "webfonts-generator": "^0.4.0",
        "tslint-microsoft-contrib": "^5.0.3",
        "crypto-js": "^3.1.9-1",
        "gulp": "4.0.0",
        "gulp-typescript": "5.0.1",
        "merge-stream": "2.0.0",
        "gulp-concat": "2.6.1",
        "gulp-header": "2.0.9",
        "gulp-replace": "1.0.0",
        "@types/react-modal": "3.1.2",
        "@types/react-tooltip": "3.3.6",
        "@types/chai": "4.0.6",
        "@types/bluebird": "2.0.33",
        "@types/fuse": "2.5.2",
        "@types/highlight.js": "9.12.2",
        "@types/jquery": "3.3.29",
        "@types/marked": "0.3.0",
        "@types/mocha": "2.2.44",
        "@types/node": "8.0.53",
        "@types/react": "16.4.7",
        "@types/react-dom": "16.0.3",
        "@types/react-redux": "5.0.0",
        "@types/request": "2.48.2",
        "@types/web-bluetooth": "0.0.4"
    },
    "scripts": {
        "build": "gulp",
        "test": "gulp travis",
        "clean": "gulp clean",
        "lint": "gulp lint",
        "start": "pxt serve", (this which i update)
        "test:all": "gulp test",
        "test:decompiler": "gulp testdecompiler",
        "test:decompiler-errors": "gulp testdecompilererrors",
        "test:err": "gulp testerr",
        "test:fmt": "gulp testfmt",
        "test:lang": "gulp testlang",
        "update": "gulp update"
    }
}

Kunal614, 2020-04-20:
[image]
I also changes the port to 8080 , what should i do in start in pacakge.json ?

pelikhan, 2020-04-20:
pxt serve if for development purposes only. Use staticpkg instead.

[Screenshot from 2020-04-20 16-50-42]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F43969265%2F79746155-28d17400-8327-11ea-8e43-a39fa3aa2a0e.png&data=02%7C01%7Cjhalleux%40microsoft.com%7C185b288f6b614bbee57808d7e51d0f93%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637229785298047240&sdata=gO%2F2mvLHrcBhK3vcZXs%2BtCSTmgKJRmZJPs%2BJFsuf7ck%3D&reserved=0>

Kunal614, 2020-04-20:
can you suggest like , what should i add?

pelikhan, 2020-04-20:
See https://makecode.com/cli/staticpkg

closed 2020-04-20 by pelikhan