microsoft/pxt-maker #311: Share button not working

gotfredsen, 2020-04-23:
Hi,

The Share button 
[image] 
does not work on our implementation of pxt-maker.

We get the following error, and I can't source the issue:
[image]

Any help will be appreciated.

pelikhan, 2020-04-23:
It is only supported for editors hosted by Microsoft. You will have to implement the share api on your backend to support this.

[Screenshot 2020-04-23 at 3 37 25 PM]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F2776277%2F80105703-39792880-8579-11ea-9d7d-145720d3fe36.png&data=02%7C01%7Cjhalleux%40microsoft.com%7C3b22dc9806dd47b2365708d7e78cc55f%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637232464102529920&sdata=7%2FLRvpkEiYFrD6FTCgQ7%2FsnSjUWcjBE1nDj%2Buy%2BQ3Ws%3D&reserved=0>

[Screenshot 2020-04-23 at 3 18 35 PM]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F2776277%2F80105843-64637c80-8579-11ea-87e9-f97d457e293e.png&data=02%7C01%7Cjhalleux%40microsoft.com%7C3b22dc9806dd47b2365708d7e78cc55f%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637232464102539913&sdata=znE9CzHYi6HpPYUQRhFm8Z8nR1JSiLL3GiNjH3368ik%3D&reserved=0>

gotfredsen, 2020-04-23:
Ok, Thanks Peli

closed 2020-04-23 by gotfredsen