microsoft/pxt-maker #318: Analog read not working on Circuit Playground Bluefruit

jedgarpark, 2020-10-20:
**Describe the bug**
None of the analog read pins seem to be working for me.

**To Reproduce**
I hooked up a 10k pot to 3v3, GND, and A3 (tried them all eventually) then ran the code below. It never changed the graph as I turned the pot.
```
forever(function () {
    light.graph(pins.A3.analogRead(), 255)
})
```
**Expected behavior**
Graph level changes as pot is turned.

**Desktop (please complete the following information):**
 - OS: mac os Catalina 10.15.7
 - Browser: Chrome 86.0.4240.80
 -Maker MakeCode Version: 
MakeCode Maker (Cortex class) version:  0.13.60

Microsoft MakeCode version:  6.4.5

codal-nrf52840-dk runtime version:  v1.1.7

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

yellowjaguar5, 2020-10-20:
Thanks, @jedgarpark for submitting this issue. I have also experienced this problem on Circuit Playground Bluefruit.

nholbert, 2022-07-21:
I've run into this issue working with kids and the Circuit Playground Bluefruit as well. Is there a newer version of MakeCode that has addressed this error and also works with the Bluefruit?