microsoft/pxt-maker #328: Adafruit grand central device doesn't read makecode program

lalauria2000, 2021-01-18:
Hi, I am working on the files for the adafruit grand central. The board is already on my local makecode. I can do a new project with it and the simulator works fine (for example when I program the neopixel to turn on as you can see in the screenshots). The problem is that when I download the .uf2 file and put it in my device it doesn't work. I've checked eveything and I can't see to find the problem. The bootloader is not a problem because I already updated it (I updated it with the adafruit v.10 first and then with the microsoft one: v.3.4, now my board is running with the microsoft one.)
I dont know if the problem has to do with the files I created for the Grand Central (which seem to work fine as the board is on the simulator) or what the problem is. 

PS: it is true that with other makecode blocks (when I program the "on button D2 click" purple block for example, some random cables and buttons appear in the simulator)
So basically my problem is: 1) my device doesnt read any makecode program and 2) the simulator does random things.

If you need the updated files of the grand central, I can attach them. 
Thank you

[image]
[image]
[image]
[image]

pelikhan, 2021-01-19:
Make sure that all the pins have been mapped properly. Check the values in ``config.ts`` against the schematics from Grand Central in Adafruit.

lalauria2000, 2021-01-19:
Just checked it again and they are all mapped okay...

wallarug, 2021-01-19:
@pelikhan  Would there be any reason that MakeCode wouldn't like the fact that there are more registers in this SAMD51 processor compared to other boards?

pelikhan, 2021-01-19:
It’s usually the clocks. Where are your PRs?

lalauria2000, 2021-01-19:
Here are all the files for the grand central if you want to check them out. Also the dal.d.ts of core---samd51
https://drive.google.com/drive/folders/1aVYAzUJynknWLzuT9VYk9fQLoQf3P3P1?usp=sharing

pelikhan, 2021-01-19:
A PR would be easier.

https://drive.google.com/drive/folders/1aVYAzUJynknWLzuT9VYk9fQLoQf3P3P1?usp=sharing<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdrive.google.com%2Fdrive%2Ffolders%2F1aVYAzUJynknWLzuT9VYk9fQLoQf3P3P1%3Fusp%3Dsharing&data=04%7C01%7Cjhalleux%40microsoft.com%7Cfefe7a2e7b0e4dd91f5a08d8bc8b8d6c%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637466654845700165%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=2ftsucqxQepmocZ%2Blvq1gUWqoZyvKIOhbfQ%2F8PFUHcM%3D&reserved=0>

lalauria2000, 2021-01-19:
just opened a new one

pelikhan, 2021-01-19:
Could you paste the link here?

lalauria2000, 2021-01-19:
https://github.com/microsoft/pxt-maker/pull/329

wallarug, 2021-03-14:
Hi there @pelikhan ,

**RE:  Adafruit Grand Central M4 Express**

I have been testing this out and have more details on the problems with the physical hardware.  I need some assistance from someone who understands the deep inner workings of MakeCode a bit better than me.

So the simulator works perfectly and as expected.  The issue is with the software being deployed to the physical board.

I bought an Adafruit GC M4 Express and wanted to see this for myself.   What happens is, no matter what program you put onto the board, it just flashes the D13/LED constantly in a set pattern.

3 very short bursts
3 short bursts

I cannot work out why the physical board is behaving this way.   

I have changed the memory allocation, changed the openocd commands in the pxtarget file, adjusted all the pins to the correct Pwm / Digital IO setting.  I've run out of ideas at this point and considered just starting from scratch using the Adafruit Metro M4 as the base board.

I suspect this behaviour has something to do with using the larger SAMD51P20A chip vs the currently supported SAMD51J19A chips.

Any ideas how to get this board up and running?

Thanks in advance.

pelikhan, 2021-03-14:
It's SOS ... --- .... So something went wrong with the config or hardware. I think we have a guide somewhere to attach gdb and read the debug buffer, panic code. @mmoskal do you remember where you keep that guide?

wallarug, 2021-03-14:
Thanks @pelikhan ,

If it's any help.  I just did another test with the board using the Metro M4 Express and it is able to run a program no problem.  Not sure if this will assist in pointing at the potential problem here.

[image]

I had an LED physically connected to A13 on the Adafruit Grand Central M4.

lalauria2000, 2021-03-23:
Hi @pelikhan @mmoskal could you share the SOS guide to attach gdb and read the debug buffer??

jamesadevine, 2021-03-23:
@lalauria2000 https://makecode.com/develop/hw-debugging

Is that what you are looking for?

lalauria2000, 2021-03-23:
@jamesadevine It says it works for Circuit Playground Express devices, should I follow those guidelines for the Adafruit Grand Central? I really have no idea if I should use that guide or there is another one...

wallarug, 2021-04-10:
Hey @jamesadevine ,

I have been also looking at this.  

Could it be because the dependency on the [itsybitsy-m4 codal](https://github.com/lancaster-university/codal-itsybitsy-m4) library is not configured to build for a SAMD51P19A?

The Adafruit Grand Central does work with the Metro M4 board.  

I'm going to keep striping back the Grand Central M4 code and swapping it with Metro M4 until something starts working.

wallarug, 2021-04-10:
Hi All,

I think I have solved this one.

In this [commit](https://github.com/microsoft/pxt-common-packages/commit/fd5a0753b498b48fd084d291cedc612618bc9619) we updated the samd file to recognise the extra PC and PD registers but we did not update the total number of pins.

```
// if we ever want to support 100+ pin packages, need to add PC,PD ports and increase this to 128
#define DEV_NUM_PINS 128
```

It is even commented in there to do it.    This was 64.

Updating it to 128 does not appear to affect the other board builds.  I tested the Metro M4 build and all seems fine.

Can we make a PR and push this change into the pxt-common-packages repo?

pelikhan, 2021-04-10:
A PR would be great!

wallarug, 2021-04-10:
No worries!  PR is in.  https://github.com/microsoft/pxt-common-packages/pull/1245

I have tested that pins on PD do work as well.

Thank you everyone for your support on this.