microsoft/pxt-maker #330: RP2040 support!

ajs256, 2021-01-24:
**Is your feature request related to a problem? Please describe.**
There should be a more noob-friendly way to program RP2040-based dev boards. 

**Describe the solution you'd like**
MakeCode Maker (or a new editor) should support RP2040-based boards. 

**Describe alternatives you've considered**
N/A

**Additional context**
The RP2040 is the first microcontroller chip from Raspberry Pi. The first board using it is the [Raspberry Pi Pico](https://rptl.io/pico), but Adafruit and SparkFun are working on their own boards. This will probably quickly become popular, and I would like to see a MakeCode editor for it.

chatelao, 2021-03-02:
I think this is a great idea and probably hundreds of RP2040 based boards will appear soon:

=> Starting with the basic edition would be great: https://www.raspberrypi.org/documentation/rp2040/getting-started/

IrregularShed, 2021-03-24:
Arduino are also developing an RP2040-based board, which shows even the longhorns of physical computing are taking the chip seriously! Unsurprisingly to anyone who knows them, Pimoroni are as well.

I'd love to see MakeCode fill the void of JS environments for this silicon. ❤️

salmanfarisvp, 2021-09-02:
If we need to add support for RP2040, what are the things need to Do/Create? Do we have any guide that helps us to provide support for a new family of microcontrollers such as RP2040? Please guide me, even if you can share the resource, that will be great. Thanks.

Vegz78, 2021-11-17:
https://github.com/microsoft/pxt-maker/search?q=rp2040&type=commits

jamesadevine, 2021-11-17:
Yes, we added RP2040 support a few weeks ago. Enjoy!

closed 2021-11-17 by jamesadevine

ChantelleJoy, 2023-03-02:
> Yes, we added RP2040 support a few weeks ago. Enjoy! [image]

FYI
The Analog Pins are not working.

I've looked into this and I think it's because GPIO 26/27 are set up in makecode as 'Analog' but only as regular Analog Pins would be, not actually set to read through the Pico's ADC hardware.

Alternatively if pins A0 / A1 were included, those could be read as Analog, automatically read through the ADC I believe.

It would be great to getting Analog Inputs working on the Pico in MakeCode - if anyone has the ability to update the files, here's the documentation from Pi:

https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#hardware_adc