microsoft/pxt-maker #331: Bluefruit crashes "OnLoud"

chatelao, 2021-01-31:
**Describe the bug**
Any sketch with "OnLoud" seams to crash the Cricuit Playground Bluefruit

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://makecode.com/_Vc1dYMCDa7rP
2. Load on CPB

**Expected behavior**
1. Animation should start first => doesn't
2. On clapping sound should play => doesn't

**Screenshots**
[image]

chatelao, 2021-02-06:
Dear @mmoskal, do you have any idea?

mmoskal, 2021-02-10:
sorry, no idea...

pelikhan, 2021-02-10:
This probably kicks the microphone code and I don't think we have it wired up.

chatelao, 2021-02-10:
Thanks for your reply, I will continue investigation. Getting a testserver running is a bit an mess.

If there would be a pre-packaged test environment for local debuging: just tell me the shortcut.

Using the Adafruit Vagrant box did not work out.