microsoft/pxt-maker #350: Code not running on  LILYGO® TTGO T8 ESP32-S2

louisvangeldrop, 2021-11-17:
**Describe the bug**
After downloading the generated .bin file, the program will not run

**To Reproduce**
Steps to reproduce the behavior:
1. I followed the instructions at adafruit to install TinyUF2. I had to put usb dip-switches in 'otg' mode
2. Download the tinyuf2-lilygo_ttgo_t8_s2_st7789-0.5.2.zip file
3. With the online ESPTool I have flashed the combined.bin at location 0x0
4. Worked well. TTGOS2BOOT (D:) pops up.
5. In maker.makecode.com; select the esp32 board
6. In Javascript:
7. forever(function () {
    pins.LED.digitalWrite(false)
    control.waitMicros(1 * 1000 * 1000)
    pins.LED.digitalWrite(true)
    control.waitMicros(1 * 1000 * 1000)
})

8. Test the program with the simulator. LED-light blinks. Works fine
9. Download the.bin file to the UF2 D-disk as "maker-espressif-esp32-devkit-c (3).bin"
10. LED on the board is not blinking.
11. Switch the usb-dipswitch to 'USB"-mode.
12. Helas. LED on the board is not blinking.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Edge
 - Version: 97

jamesadevine, 2021-11-17:
Unless the board is pin compatible with an existing target board, it is unlikely that a MakeCode program will work. Here is an example of a pin mapping for one of our internal ESP32-S2 boards: https://github.com/microsoft/pxt-maker/blob/master/libs/jacdac-iot-s2/config.ts You will need to contribute something similar for your particular target board. 

If it is the case your board is pin compatible with an existing target board, then we can have a look to see if there is an underlying issue.

louisvangeldrop, 2021-11-17:
I tried with Arduino using ESP32S2 native USB using an LED connected to pin 3 and that works well. Unfortunately when I do the same with Makecode it still does not work

louisvangeldrop, 2021-11-21:
The ESP32S2-board I am using is pin compatible with the espressif-esp32-devkit-c board. The download-button creates a .bin file. Can the extension cause the problem?

louisvangeldrop, 2021-12-11:
How do I flash the bin-file to a ESP32-WROVER-E board?

louisvangeldrop, 2023-01-31:
Flashing using this procedure:

1. Use https://espressif.github.io/esptool-js/ as online flash tool
2. 0x1000   bootloader.bin
3. 0x8000   partitions.bin. I used the bootloader and partiitions files from the espruino esp32 binaries folder
4. 0x10000 makecode.bin. The file that is generated by Makecode