microsoft/pxt-maker #352: RP2040 - pairing Mode problem

dsssssssss9, 2021-12-01:
**Describe the bug**
In Chromium browser board not available when you try to pair.

Have to manually put board into bootloader mode & copy UF2 file to it

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'Connect Device
2. Click on '....'Connect Device
3. No compatible devices found

**Expected behavior**
Should be able to select the RP2040 to pair

**Screenshots**
Add screenshots to help explain your problem. You can copy paste the screenshot in the github report. The .gif screen recording is very useful as well. 

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Browser Chromium Version 96.0.4664.45 

**Additional context**
If i try using Brave browser i can seem to pair but when i try to download code it takes quite a while & then the LED on the RP2040 starts flashing S.O.S & Makecode goes to the save UF2 manually screen

bance, 2022-08-17:
same here. webusb pairing doesn't work for rp2040 target. pico doesn't show up in the pairing selection window.

deckeye, 2024-11-28:
Once I had to download the .sf2 file and put it directly into the Raspberry Pi Pico's storage.