microsoft/pxt-maker #367: RP2040 Pico-w support

louisvangeldrop, 2022-10-16:
Would it be nice to have the RP2040 Pico-w supported in Makecode. The mcu supports Wifi  and this  allows  MQTT-based data exchange with all kinds of Home Automation packages.

ErinMutchler, 2022-10-21:
I agree. The RP2040 Pico doesn't currently work with analog writing or reading though either so I am not there are plans to get to this unfortunately.

louisvangeldrop, 2022-10-23:
www.moddable.com and kalumajs.org do offer Javascript support for the RP2040 Pico-w board. Moddable offers full javascript support and also offers the best continuity.
The strength of Makecode and Kaluma is that you can run in the browser. You don't need to install all the hardware related SDK's