microsoft/pxt-maker #378: ESP32-S3

samuk, 2024-04-22:
Is this project [abandoned?](https://github.com/microsoft/pxt-maker/graphs/contributors) Would anyone merge a pull?

campey, 2024-09-11:
@pelikhan, Allan, @tballmsft, @mmoskal and I had a call the other day for Allan and I to start maintaining pxt-maker with a view to getting more-common chips working again, as many of the chips in the project are less-used as it was largely a dumping ground for experiments.

My focus is the pi pico RP2040, ESP32 would also make a lot of sense.

The MS research team are still owners said If we're willing to do the work they will merge it. So if you're keen to collab, that'd be cool!

TannerGabriel, 2024-12-28:
@campey Any progress? I would also be interested in fixing the ESP32 and RP2040 projects.

jarmitage, 2025-02-12:
@campey also very interested in ESP32, any updates?

campey, 2025-03-28:
@TannerGabriel I got it working on my machine (TM) and now need to finish off some of the pin mappings, tidy up the md a bit, figure out what to do with node hell and get it deployed.

This is what it is looking like.
https://www.instagram.com/reel/DFmnbQjMcjs/?utm_source=ig_web_copy_link&igsh=MzRlODBiNWFlZA==

campey, 2025-03-28:
@jarmitage my friend and colleague at afrolabs @BenjaminAfrolabs is an ESP32 fan and will likely be picking up on similar work on the ESP32, my ESP32 is limited.

jarmitage, 2025-03-28:
Nice work! Still very interested, if you and @BenjaminAfrolabs are able to share any updates later on